cfriedt/greybus-for-zephyr #65: Connection diagram

anujdeshpande, 2021-03-08:
[image]
Hey @cfriedt 
I am trying to wire up my Linux machine to a CC1352 Launch board. I start gbridge with `sudo ./gbridge -p /dev/ttyS0 -b 115200` where S0 is the port that I have connected to the DIO12/DIO13 pins of CC1352 launch board. Along with GND. Are there any other pins that I am supposed to wire up? Running the UART example

anujdeshpande, 2021-03-08:
One thing that is bugging me is that in the [docs for CC1352 on zephyr](https://docs.zephyrproject.org/latest/boards/arm/cc1352r1_launchxl/doc/index.html) there's only UART0 - but the board config in the sample for greybus points to UART_1. Does that work fine?

cfriedt, 2021-03-08:
It looks as though uart_1 isn't enabled. I would suggest going through menuconfig to see how to enable it, or possible look through the generated config under `build/`

Fwir, @vaishnav98 got the UART demo up to speed again, but maybe there was some other config item needed.

UART1 is brought out to some pins on one of the Mikrobus headers.

vaishnav98, 2021-03-08:
Hi @anujdeshpande , It looks like the uart_1 is not enabled by default as @cfriedt suggested, this might be since during testing the changes were made in the zephyr  boards/arm/cc1352r1_launchxl/cc1352r1_launchxl.dts to enable UART1, the changes you will need to make in the dts are these additions in cc1352r1_launchxl.dts:
```
define IOC_PORT_MCU_UART1_TX 0x00000014
#define IOC_PORT_MCU_UART1_RX 0x00000013

&uart1 {
    status = "okay";
    current-speed = <115200>;
    tx-pin = <24 IOC_PORT_MCU_UART1_TX>;
    rx-pin = <22 IOC_PORT_MCU_UART1_RX>;
};
/* DIO22/DIO24 RX/TX */
```
Another easier solution is to disable logging and console from UART0 and use UART0 for transport, also for reference here is a modified DTS which is modified for using UART0(XDS connected UART when jumpers are in place) for transport and UART1 for logging/console

cfriedt, 2021-03-09:
> ```
> define IOC_PORT_MCU_UART1_TX 0x00000014
> #define IOC_PORT_MCU_UART1_RX 0x00000013
> 
> 
> &uart1 {
>     status = "okay";
>     current-speed = <115200>;
>     tx-pin = <24 IOC_PORT_MCU_UART1_TX>;
>     rx-pin = <22 IOC_PORT_MCU_UART1_RX>;
> };
> /* DIO22/DIO24 RX/TX */
> ```

That's right! Thanks @vaishnav98 - I almost forgot about that. I guess one change we could make to the uart example would be to define the uart1 macros in our dts overlay. Should compile ok, no? It's just sample code.

> Another easier solution is to disable logging and console from UART0 and use UART0 for transport, also for reference [...]

I've also swapped uart0 and uart1 so that uart1 is used for logging (and is just not connected in the case that logging is unimportant).

anujdeshpande, 2021-03-09:
Thanks for the help folks! Got it working as expected with the addition to the device tree. 

Would love to send a PR documenting how to go about getting UART to work - in case there's a recommended way of getting things working

closed 2021-04-21 by anujdeshpande