cfriedt/greybus-for-zephyr #66: subsys: greybus: can adc be supported?

anujdeshpande, 2021-03-09:
Are there plans to add ADC support? I see in the Greybus [spec](https://raw.githubusercontent.com/cfriedt/greybus-for-zephyr/master/doc/GreybusSpecification.pdf), in `Chapter 11 Bridged PHY Connection Protocols` that there is no mention of ADC. Is this something that would first be added to Greybus on the host, before considering that here?

cfriedt, 2021-05-13:
It doesn't seem like there is currently any ADC support in the spec, but I agree - that would be super useful. Have you asked about it on the mailing list?

anujdeshpande, 2021-05-27:
@cfriedt https://lists.linaro.org/pipermail/greybus-dev/2021-May/001376.html

Greg said he'd be open to accepting patches around this - but there's no recommended way to go about this at the moment

cfriedt, 2021-05-27:
@anujdeshpande - it looks as though there are occasional Greybus patches for Linux on the Greybus mailing list.

Based on Greg's feedback, I would say
1. Add a device class in Linux for ADC's. This will likely be the most difficult part, as there are a huge range of options and capabilities to look at to cover various vendor silicon. Hopefully a lot can be borrowed from IIO. This may take a very long time, so by all means use your prototype in production.
1. Propose a change to the Greybus spec accompanied by a proof of concept for the upstream project ara / nuttx code as well as some tests (tests could use gbsim, but you might be able to find a more creative solution). The spec / protocol changes should be simple enough that every ADC could share a common interface. Again, processes take time, so by all means use your prototype in production.
1. Would be great if you could eventually make a PR here 😉

It's really exciting to see some action happening around Greybus. I'm tempted to make some t-shirts and swag, free of course to those who can show git commit hashes!!