cfriedt/greybus-for-zephyr #67: Issue linking ring buffers for new board support

anujdeshpande, 2021-04-05:
Hi @cfriedt @vaishnav98 

I am trying to add support for UART transport, and ran into some issues 

```
arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m+fp/hard  -specs=nano.specs && :
/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: modules/greybus/subsys/greybus/lib..__greybus__subsys__greybus.a(transport-uart.c.obj): in function `uart_work_fn':
/home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:57: undefined reference to `ring_buf_get'
/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:82: undefined reference to `ring_buf_get'
/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: modules/greybus/subsys/greybus/lib..__greybus__subsys__greybus.a(transport-uart.c.obj): in function `gb_xport_uart_isr':
/home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:195: undefined reference to `ring_buf_get'
/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:204: undefined reference to `ring_buf_put'
```

I saw this with both `stm32f3_disco` and `lpcxpresso55s69_cpu0`. This was after I added the requisite board conf and overlay files.

I am trying to understand what these chipsets need to support in order for UART to work. One thing I did realize that both these chips have usart and not uart in their device trees under Zephyr. Could that be a problem when adding support for them?

cfriedt, 2021-04-05:
Interesting. I have a couple of disco boards, so perhaps I can check it out.. will take me a day or so to look into it.

anujdeshpande, 2021-04-05:
@cfriedt i pushed a branch for you here - https://github.com/anujdeshpande/greybus-for-zephyr/tree/add-stf303-uart

If it's simpler I can open a PR here. I don't think you'll need a board tbh - since this seems an issue when building.

anujdeshpande, 2021-04-05:
Full logs in case it helps -

```
(base) ➜ zephyr (fc1b5de4c3) ✗ west build -b $BOARD ../greybus/samples/subsys/greybus/uart
-- west build: generating a build system
-- Application: /home/anuj/zephyrproject/greybus/samples/subsys/greybus/uart
-- Zephyr version: 2.4.99 (/home/anuj/zephyrproject/zephyr)
-- Found Python3: /home/anuj/miniconda3/bin/python3.7 (found suitable exact version "3.7.6") found components: Interpreter
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: stm32f3_disco
-- Cache files will be written to: /home/anuj/.cache/zephyr
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/home/anuj/zephyr-sdk-0.12.3)
-- Found dtc: /home/anuj/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/anuj/zephyrproject/zephyr/boards/arm/stm32f3_disco/stm32f3_disco.dts
-- Found devicetree overlay: /home/anuj/zephyrproject/greybus/samples/subsys/greybus/uart/boards/stm32f3_disco.overlay
-- Generated zephyr.dts: /home/anuj/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/anuj/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/anuj/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/anuj/zephyrproject/zephyr/boards/arm/stm32f3_disco/stm32f3_disco_defconfig'
Merged configuration '/home/anuj/zephyrproject/greybus/samples/subsys/greybus/uart/prj.conf'
Merged configuration '/home/anuj/zephyrproject/greybus/samples/subsys/greybus/uart/boards/stm32f3_disco.conf'
Configuration saved to '/home/anuj/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/anuj/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anuj/zephyrproject/zephyr/build
-- west build: building application
[1/163] Preparing syscall dependency handling

[125/163] Building C object modules/greybus/subsys/greybus...greybus__subsys__greybus.dir/platform/transport-uart.c.obj
/home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c: In function 'gb_xport_send':
/home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:149:18: warning: passing argument 1 of 'sendMessage' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  149 |  r = sendMessage(uart_dev, msg);
      |                  ^~~~~~~~
/home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:102:39: note: expected 'struct device *' but argument is of type 'const struct device *'
  102 | static int sendMessage(struct device *dev, struct gb_operation_hdr *msg)
      |                        ~~~~~~~~~~~~~~~^~~
[130/163] Building C object modules/greybus/subsys/greybus...eFiles/..__greybus__subsys__greybus.dir/platform/i2c.c.obj
/home/anuj/zephyrproject/greybus/subsys/greybus/platform/i2c.c:35:12: warning: 'greybus_i2c_control_init' defined but not used [-Wunused-function]
   35 | static int greybus_i2c_control_init(const struct device *dev) {
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
[158/163] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf
: && ccache /home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=/home/anuj/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/serial/libdrivers__serial.a  modules/greybus/subsys/greybus/lib..__greybus__subsys__greybus.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj  -L"/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/thumb/v7e-m+fp/hard"  -L/home/anuj/zephyrproject/zephyr/build/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfpu=fpv4-sp-d16  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lm  -lc  -lgcc  -L"/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m+fp/hard  -specs=nano.specs && :
/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: modules/greybus/subsys/greybus/lib..__greybus__subsys__greybus.a(transport-uart.c.obj): in function `uart_work_fn':
[... cut ...]
/home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/anuj/zephyrproject/greybus/subsys/greybus/platform/transport-uart.c:204: undefined reference to `ring_buf_put'
Memory region         Used Size  Region Size  %age Used
           FLASH:       48248 B       256 KB     18.41%
            SRAM:       31392 B        40 KB     76.64%
        IDT_LIST:         216 B         2 KB     10.55%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/anuj/zephyrproject/zephyr/build
```

anujdeshpande, 2021-04-07:
@cfriedt did you get time to check into this? Let me know if there's some more info that I can provide

anujdeshpande, 2021-04-08:
I was digging through this more - 

build-failure.log
```
cmake version 3.16.3 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake --build /home/anuj/zephyrproject/zephyr/build -- -v
[1/163] cd /home/anuj/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
```

build-success.log
```
cmake version 3.16.3 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake --build /home/anuj/zephyrproject/zephyr/build -- -v
[1/171] cd /home/anuj/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
```

So there's a bunch of tasks that cmake isn't picking up on. I was digging more into what it missed, and I couldn't find this in the one that fails

```
[34/171] ccache /home/anuj/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.4.0-3009-gfc1b5de4c307 -DKERNEL -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/ti_simplelink/cc13x2_cc26x2 -I../lib/libc/newlib/include -I/home/anuj/zephyrproject/greybus/include -I/home/anuj/zephyrproject/greybus/subsys/greybus/../../../include -I/home/anuj/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/anuj/zephyrproject/modules/hal/ti/simplelink/source/ti/devices/cc13x2_cc26x2/. -I/home/anuj/zephyrproject/modules/hal/ti/simplelink/. -I/home/anuj/zephyrproject/modules/hal/ti/simplelink/source -I/home/anuj/zephyrproject/greybus/samples/subsys/greybus/uart/../../../../../include -Os -imacros /home/anuj/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/anuj/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/anuj/zephyrproject/greybus/samples/subsys/greybus/uart=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/anuj/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/anuj/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=c99 -MD -MT zephyr/CMakeFiles/zephyr.dir/lib/os/ring_buffer.c.obj -MF zephyr/CMakeFiles/zephyr.dir/lib/os/ring_buffer.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/lib/os/ring_buffer.c.obj   -c /home/anuj/zephyrproject/zephyr/lib/os/ring_buffer.c
```

The only thing that's done before this cmake, is the device tree stuff I believe. So this seems to be coming from that. Will keep digging

[build-failure.log](https://github.com/cfriedt/greybus-for-zephyr/files/6278151/build-failure.log)
[build-success.log](https://github.com/cfriedt/greybus-for-zephyr/files/6278152/build-success.log)

anujdeshpande, 2021-04-14:
@vaishnav98 or @cfriedt - do you know if the greybus uart transport cares if the peripheral is `usart` or `uart`.

cfriedt, 2021-04-14:
@anujdeshpande - I would guess that `CONFIG_RING_BUFFER` is not set. Try this:

```bash
grep RING build/zephyr/.config                
# CONFIG_RING_BUFFER is not set
```

I see it's not in `samples/subsys/greybus/uart/prj.conf`, so if you're making a PR, please add it there.

cfriedt, 2021-04-14:
Going to give it a quick shot now. Been very busy in the last couple of weeks

anujdeshpande, 2021-04-14:
@cfriedt that fixed it! I didn't realize that we needed to add that config to the cmake. I can build without errors now. Will try it out on hardware and send a PR!

anujdeshpande, 2021-04-14:
I'm curious tho - Why didn't the Nordic or TI boards need this flag tho? How were they picking this up by default

cfriedt, 2021-04-14:
It's quite likely that one of the UART or wireless drivers already pulls in ring buffer support on those boards.

cfriedt, 2021-05-13:
@anujdeshpande - were you able to resolve this one? Did you make a PR?

anujdeshpande, 2021-05-18:
@cfriedt sent PR! Apologies, forgot about pushing this upstream!

PR #68 (2021-06-08): Add config for including ring buffer
Fixes #67 

Change-type: minor
closed 2021-06-08 by cfriedt